Portisch/RF-Bridge-EFM8BB1 #177: DC1602 remote (Dooys) similar / same as DC90 not read the first byte correctly

nosaj66au, 2020-10-29:
### Latest firmware used?
Be sure you tried the [latest release](https://github.com/Portisch/RF-Bridge-EFM8BB1/releases) or the [newest binary](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex) of the firmware!
AA0355
### Information
What device you try to sniff data from? Description? Brand and type?
DC1602 (Dooya)
### Sniffed data
Post your sniffed data by command [0xB1](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xB1) or [0xA6](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA6).

"{"RfRaw":{"Data":"AA B1 05 12CA 05F0 015E 02E4 2224 481A3A3B2B2B2A3B2A3A3A3B2B2A3B2A3B2A3B2A3A3B2B2B2A3A3B2A3B2B2A3B2B2A3A3A3B2B2B2B2A 55"}}" or 

"{"RfRaw":{"Data":"AAA606013A354E501155"}}"  > 90% of the time
"{"RfRaw":{"Data":"AAA6060B3A354E501155"}}" > 10% of the time  

I have determined that B3A354E is the correct ID for my remote and the correct protocol is DC90 (08) when I transmit i have 100% success using rfraw AAA8080B3A354E501155 to control all blind functions of all blinds. This done by correctly altering the last 5 digits. All Good.

However when using rfraw 166 only about 10% of the time does it return the correct device id returning a 0x1 instead of 0xB.
I think is is because it is detecting protocol 01 which has slightly different timing but is close enough to work sometimes.
Is the a way to fix the protocol detection without recompiling ?

Jason

sthorley, 2020-11-12:
Hey mate, This project seems kinda dead so I doubt anyone's able to help you much.

You say that you have a Dooya 1602 working 90% of the time and you managed to figure out the protocol? Would you mind sharing the codes you sniffed? I've been able to capture codes using rfraw 177 and after converting them to B0 I've had no luck getting them to work :( 

These were my 177 captures for the Up button:
{"RfRaw":{"Data":"AA B1 04 136A 0262 0118 1374 38192A192A1A1A19292A1929292A19292A19292A19292A1A1929292A192A1A1A1A192929292929292929292929292929292A1A1A1A192A19292A1A192A192A1A1A19 55"}}
{"RfRaw":{"Data":"AA B1 04 137E 0262 0118 137E 38192A192A1A1A19292A1929292A19292A19292A19292A1A1929292A192A1A1A1A192929292929292929292929292929292A1A192A1A192A1A1A192A1A1A1A1A1929 55"}}

And the corresponding BO codes:
AA B0 4C 04 08 136A 0262 0118 1374 38192A192A1A1A19292A1929292A19292A19292A19292A1A1929292A192A1A1A1A192929292929292929292929292929292A1A1A1A192A19292A1A192A192A1A1A19 55
AA B0 4C 04 08 137E 0262 0118 137E 38192A192A1A1A19292A1929292A19292A19292A19292A1A1929292A192A1A1A1A192929292929292929292929292929292A1A192A1A192A1A1A192A1A1A1A1A1929 55

nosaj66au, 2020-12-02:
> Hey mate, This project seems kinda dead so I doubt anyone's able to help you much.
> 
> You say that you have a Dooya 1602 working 90% of the time and you managed to figure out the protocol? Would you mind sharing the codes you sniffed? I've been able to capture codes using rfraw 177 and after converting them to B0 I've had no luck getting them to work :(
> 
> ~These were my 177 captures for the Up button: {"RfRaw":{"Data":"AA B1 04 136A 0262 0118 1374 38192A192A1A1A19292A1929292A19292A19292A19292A1A1929292A192A1A1A1A192929292929292929292929292929292A1A1A1A192A19292A1A192A192A1A1A19 55"}} {"RfRaw":{"Data":"AA B1 04 137E 0262 0118 137E 38192A192A1A1A19292A1929292A19292A19292A19292A1A1929292A192A1A1A1A192929292929292929292929292929292A1A192A1A192A1A1A192A1A1A1A1A1929 55"}}~
> 
> ~And the corresponding BO codes: AA B0 4C 04 08 136A 0262 0118 1374 38192A192A1A1A19292A1929292A19292A19292A19292A1A1929292A192A1A1A1A192929292929292929292929292929292A1A1A1A192A19292A1A192A192A1A1A19 55 AA B0 4C 04 08 137E 0262 0118 137E 38192A192A1A1A19292A1929292A19292A19292A19292A1A1929292A192A1A1A1A192929292929292929292929292929292A1A192A1A192A1A1A192A1A1A1A1A1929 55~
> 
> I now have a working config see #179

Hi sthorley, You can see what I was getting above. Only got the correct message 10% of the time. Hook up my DSCRO the an RF receiver module and confirmed datagram. In my testing only the 8th byte was ever wrong.
As I was linking my blinds to Node Red and the RF bridge I just found the ID that worked (B3A354E for this blind). All the other bytes are just "blind number" and commands (Up, Down, Stop and Program)
Hope this helps
Jason