Portisch/RF-Bridge-EFM8BB1 #182: Duplicate commands when sending B0 signal to SZ Funpower device

zyrorl, 2020-11-25:
### Latest firmware used?
Be sure you tried the [latest release](https://github.com/Portisch/RF-Bridge-EFM8BB1/releases) or the [newest binary](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex) of the firmware!

Yes.

### Information
What device you try to sniff data from? Description? Brand and type?
SZ Funpower Remote (branded as brilliant lighting), manufactured on 2020-06-22 
the model seems to be FT1211R-V2

I have 3 other older remotes, similar to this one (also SZ Funpower), which all work fine sniffing the B1 codes and converting to B0. But this new revision of the remote appears to somehow behave weirdly.

When I press the remote button, the sniffed B1 code converted to B0 appears to trigger the device to enact the command twice (for example the ceiling fan receives the power on/off toggle twice so it stays on or off).

### Sniffed data
Post your sniffed data by command [0xB1](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xB1) or [0xA6](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA6).

4 Subsequent commands for Power on.

04:44:42 CMD: rfraw 177
04:44:42 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
04:44:42 MQT: tele/sonoff/RESULT = {"Time":"2020-11-25T04:44:42","RfRaw":{"Data":"AAA055"}}
04:44:47 MQT: tele/sonoff/RESULT = {"Time":"2020-11-25T04:44:47","RfRaw":{"Data":"AA B1 03 0136 037A 164E 28181908190819081818190908190908181818181818190908 55"}}
04:44:50 MQT: tele/sonoff/RESULT = {"Time":"2020-11-25T04:44:50","RfRaw":{"Data":"AA B1 03 012C 037A 1644 28181908190819081818190908190908181818181818190908 55"}}
04:44:52 MQT: tele/sonoff/RESULT = {"Time":"2020-11-25T04:44:52","RfRaw":{"Data":"AA B1 03 0136 037A 1644 28181908190819081818190908190908181818181818190908 55"}}
04:44:54 MQT: tele/sonoff/RESULT = {"Time":"2020-11-25T04:44:54","RfRaw":{"Data":"AA B1 03 0136 037A 1644 28181908190819081818190908190908181818181818190908 55"}}

I'm using https://bbconv.hrbl.pl/ to convert the codes.

Shadowpheonix, 2021-08-24:
I am having this same issue.   If I press the button on the remote once, the light will beep once and perform the command.  If I send the decoded B0 command from the Sonoff device, the light beeps twice and acts as if the command was sent twice.

B1 results for the Light On command...
09:22:16.320 CMD: rfraw 0xB1
09:22:16.327 MQT: stat/tasmota_SonoffBridge_1/RESULT = {"RfRaw":"ON"}
09:22:16.393 MQT: tele/tasmota_SonoffBridge_1/RESULT = {"Time":"2021-08-24T09:22:16","RfRaw":{"Data":"AAA055"}}
09:22:19.537 MQT: tele/tasmota_SonoffBridge_1/RESULT = {"Time":"2021-08-24T09:22:19","RfRaw":{"Data":"AA B1 04 0370 014A 1400 0384 381929 55"}}
09:22:19.574 MQT: tele/tasmota_SonoffBridge_1/RESULT = {"Time":"2021-08-24T09:22:19","RfRaw":{"Data":"AA B1 04 0348 0168 00F0 03C0 381A0A 55"}}
09:22:19.625 MQT: tele/tasmota_SonoffBridge_1/RESULT = {"Time":"2021-08-24T09:22:19","RfRaw":{"Data":"AA B1 03 0370 012C 0384 281909 55"}}
09:22:20.665 MQT: tele/tasmota_SonoffBridge_1/RESULT = {"Time":"2021-08-24T09:22:20","RfRaw":{"Data":"AA B1 03 0118 0398 1414 2818190819090908181908190819090818181819081909090908181819090819090818181909081908 55"}}
09:22:21.469 MQT: tele/tasmota_SonoffBridge_1/RESULT = {"Time":"2021-08-24T09:22:21","RfRaw":{"Data":"AA B1 03 0136 0370 13D8 2818190819090908181908190819090818181819081909090908181819090819090818181909081908 55"}}
09:22:22.273 MQT: tele/tasmota_SonoffBridge_1/RESULT = {"Time":"2021-08-24T09:22:22","RfRaw":{"Data":"AA B1 03 0140 0370 13D8 2818190819090908181908190819090818181819081909090908181819090819090818181909081908 55"}}
09:22:23.077 MQT: tele/tasmota_SonoffBridge_1/RESULT = {"Time":"2021-08-24T09:22:23","RfRaw":{"Data":"AA B1 03 0136 0366 13E2 2818190819090908181908190819090818181819081909090908181819090819090818181909081908 55"}}
09:22:23.881 MQT: tele/tasmota_SonoffBridge_1/RESULT = {"Time":"2021-08-24T09:22:23","RfRaw":{"Data":"AA B1 03 0140 0370 13D8 2818190819090908181908190819090818181819081909090908181819090819090818181909081908 55"}}

zyrorl, 2023-03-16:
any updates on this? @Portisch ?