Portisch/RF-Bridge-EFM8BB1 #184: Successful sniffing but received doesn't respond. Any suggestions?

davida72, 2020-12-04:
Sonoff RF Bridge flashed with latest Tasmota and Portisch. I'm trying to control a [smoke machine](https://www.amazon.co.uk/Machine-Wireless-Control-Halloween-Wedding/dp/B07JDBM21R/) which uses (what I assume is) an RF remote. Here's some sample sniffing.

```
17:18:30 RSL: tele/tasmota_EE27B7/RESULT = {"Time":"2020-12-04T17:18:30","RfRaw":{"Data":"AA B1 03 012C 03B6 0410 28091809 55"}}
17:18:32 RSL: tele/tasmota_EE27B7/RESULT = {"Time":"2020-12-04T17:18:32","RfRaw":{"Data":"AA B1 03 015E 03A2 2788 A0918090909180918091809180918091809090909181809090 55"}}
17:18:33 RSL: tele/tasmota_EE27B7/RESULT = {"Time":"2020-12-04T17:18:33","RfRaw":{"Data":"AA B1 04 010E 01AE 0352 0424 381A280A 55"}}
17:18:33 RSL: tele/tasmota_EE27B7/RESULT = {"Time":"2020-12-04T17:18:33","RfRaw":{"Data":"AA B1 04 00F0 01AE 037A 0438 381A281A 55"}}
17:18:34 RSL: tele/tasmota_EE27B7/RESULT = {"Time":"2020-12-04T17:18:34","RfRaw":{"Data":"AA B1 03 010E 0398 0410 28091809 55"}}
17:18:36 RSL: tele/tasmota_EE27B7/RESULT = {"Time":"2020-12-04T17:18:36","RfRaw":{"Data":"AA B1 03 0122 03CA 044C 28091809 55"}}
17:18:37 RSL: tele/tasmota_EE27B7/RESULT = {"Time":"2020-12-04T17:18:37","RfRaw":{"Data":"AA B1 04 010E 03AC 01B8 042E 38091829 55"}}
17:18:38 RSL: tele/tasmota_EE27B7/RESULT = {"Time":"2020-12-04T17:18:38","RfRaw":{"Data":"AA B1 03 012C 03DE 0410 28091809 55"}}
```

I'm using BitBucket to convert and have been sending the result.

```
17:20:51 CMD: rfraw AA B0 21 03 08 015E 03A2 2788 A0918090909180918091809180918091809090909181809090 55
17:20:51 RSL: stat/tasmota_EE27B7/RESULT = {"RfRaw":"ON"}
17:20:51 RSL: tele/tasmota_EE27B7/RESULT = {"Time":"2020-12-04T17:20:51","RfRaw":{"Data":"AAA055"}}
```

But nothing happens. I've tried with every line I've sniffed, but nothing happens. Anyone know what's going on?

OptimusGREEN, 2020-12-04:
Just spent ages with same results as you. no idea whats happening.

sthorley, 2020-12-07:
Most likely the first 6 segments will be constant (that’s what defines the protocol) so try gatherings many more rfraw results and then selecting a combination of the most commonly appearing codes. Eg: 010E appears 3 times in your sample so try building a code with that.

davida72, 2020-12-07:
Thank you. I'm going to try again and report back.

davida72, 2020-12-23:
Turns out it was an error sitting between my chair and the keyboard.

The remote I was trying to sniff was 315mhz, which the Sonoff RF Bridge does not detect. I'm going to build an ESP8266 ESPHome device to achieve something similar. I guess what I was receiving was just background noise.

closed 2020-12-30 by davida72