Portisch/RF-Bridge-EFM8BB1 #189: How to send raw B0 codes via MQTT

patrickjane, 2021-01-25:
Hi,

sorry for making this a new issue, but I can't for the life of me figure out how to send raw commands via MQTT. The documentation just is not clear to me, or maybe I'm just too stupid.

What I have & what I want to do:
I try to mimic a rf remote for roller blinds. I was able to flash the portisch firmware, and enabling the raw mode, I was able to sniff the B1 code, and eventually convert them into B0 codes. **Although** the tool spit out errors, I was able to paste those codes into the Tasmota web GUI, prefixed with "RFRAW", which actually made the bridge send out the command, and indeed my roller blind moves up and down.

Since "RFRAW" is also the command for enabling certain raw modes (there are several, if I understand this correctly), I am not sure if this is the correct thing to do, I just found this on some other github issue.
However, what I **actually* want to do, is send those commands (which already proved to be working) via MQTT, so I can integrate this into my home automation (home assistant). 

When I receive stuff from the bridge, in MQTT it looks like this:

```
tasmota/discovery/DC4F22DB5FEF/sensors {"sn":{"Time":"2020-12-23T13:57:30"},"ver":1}
tele/tasmota_DB5FEF/RESULT {"Time":"2021-01-25T17:27:30","RfRaw":{"Data":"AA B1 03 0104 026C 026C 281908 55"}}
tele/tasmota_DB5FEF/RESULT {"Time":"2021-01-25T17:27:30","RfRaw":{"Data":"AA B1 04 0208 015E 00BE 02A8 381A081A 55"}}
tele/tasmota_DB5FEF/RESULT {"Time":"2021-01-25T17:27:30","RfRaw":{"Data":"AA B1 03 0104 026C 026C 281908 55"}}
tele/tasmota_DB5FEF/RESULT {"Time":"2021-01-25T17:27:33","RfRaw":{"Data":"AA B1 03 01F4 00B4 02C6 280909 55"}}
tele/tasmota_DB5FEF/STATE {"Time":"2021-01-25T17:28:42","Uptime":"33T03:31:26","UptimeSec":2863886,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":5,"Wifi":{"AP":1,"SSId":"unity","BSSId":"FE:EC:DA:B1:BB:C4","Channel":11,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:03"}}
```

What topic and payload do I have to use to send the B0 commands? Which command do I have to use?

Sidequestions:
- do I need to *disable* raw mode when I'm done with all this?
- if so, how can I do this?

Thanks in advance.

### Latest firmware used?
Yep.

### Information
Sonoff Rf Bridge 433

### Sniffed data

```
Wildfire:Downloads s710$ python3 BitBucketConverter.py -f content.txt
###### Processing line {"RfRaw":{"Data":"AA B1 04 13B0 0280 0118 13BA 38192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A19292929292A192A192A19 55"}}
 ######
Filtered 0xB1 data: AAB10413B00280011813BA38192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A19292929292A192A192A1955
Failed to decode buckets...
Resulting 0xB0 data: AA B0 4C 04 04 13B0 0280 0118 13BA 938192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A19292929292A192A192A1 55
###### Processing line {"RfRaw":{"Data":"AA B1 04 1392 0294 012C 13C4 38192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A19292929292A192A192A19 55"}}
 ######
Filtered 0xB1 data: AAB10413920294012C13C438192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A19292929292A192A192A1955
Failed to decode buckets...
Resulting 0xB0 data: AA B0 4C 04 04 1392 0294 012C 13C4 938192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A19292929292A192A192A1 55
###### Skipping line '
' ######
###### Processing line {"RfRaw":{"Data":"AA B1 04 13B0 0276 0118 139C 38192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A19292A1A192A1A1A192A19 55"}}
 ######
Filtered 0xB1 data: AAB10413B002760118139C38192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A19292A1A192A1A1A192A1955
Failed to decode buckets...
Resulting 0xB0 data: AA B0 4C 04 04 13B0 0276 0118 139C 938192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A19292A1A192A1A1A192A1 55
```

peteakalad, 2021-01-25:
You should be able to use backlog like so, replace the middle part of the topic as appropriate for your device name:

Topic=cmnd/sonoff-rf-bridge/backlog
Payload=rfraw AA B0 4C 04 04 13B0 0276 0118 139C 938192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A19292A1A192A1A1A192A1 55; rfraw 0

This will disable rfraw once its done.

patrickjane, 2021-01-26:
Okay, so this means, I need the raw mode ONLY when I need to sniff new codes, but once set up, I can disable it, and still be able to send the B0 codes as shown above?

peteakalad, 2021-01-26:
That backlog command will temporarily enable rfraw to send the rf signal and then disables rfraw mode via the last ‘rfraw 0’ command.

patrickjane, 2021-01-26:
Ok, understood. It actually worked, I can control it just fine, so that issue is out of the way.

However, I am still confused about the raw-mode. When turned on, I don't receive the "normal" signals anymore via MQTT, e.g. `{"RfReceived":{"Sync":14050,"Low":480,"High":1380,"Data":"E2BC0E","RfKey":"None"}}`, but I guess thats the point of the raw mode.

However, in my setup, I have 3 different kinds of devices:
1. remote control for which I sniffed the B1 codes in raw mode, now using your method via MQTT to issue remote commands
2. door/window contacts, which worked fine without raw mode, e.g. I got `{"RfReceived":{"Sync":14050,"Low":480,"High":1380,"Data":"E2BC0E","RfKey":"None"}}`. Now, I get this:

(window open)
```
{"RfRaw":{"Data":"AA B1 03 056E 01A4 3796 28181819090908190819081818181909090909090819081909 55"}}
{"RfRaw":{"Data":"AA B1 03 058C 019A 3674 28181819090908190819081818181909090909090819081909 55"}}
```

(window closed)
```
{"RfRaw":{"Data":"AA B1 03 056E 01B8 36D8 28181819090908190819081818181909090909090818181909 55"}}
{"RfRaw":{"Data":"AA B1 03 0582 01AE 3674 28181819090908190819081818181909090909090818181909 55"}}
```
3. I have wall-switches with 433 radio signals, which I **don't** receive unless in raw mode. When in raw mode, I get something like:
```
{"RfRaw":{"Data":"AA B1 03 012C 0532 051E 280818 55"}}
{"RfRaw":{"Data":"AA B1 05 015E 0AF0 04E2 00FA 290E 481828383828382838282838280838282808082828083828382838282808280808283828280828080828280828382808082828082808082828082808082808282808 55"}}
{"RfRaw":{"Data":"AA B1 03 0140 050A 050A 281808 55"}}
{"RfRaw":{"Data":"AA B1 05 0168 0ADC 04C4 00E6 28C8 481828383828382838282838283838282808382828083828382838282808283838283828280828083828280828082808382828382808382828082808382838282838 55"}}
{"RfRaw":{"Data":"AA B1 03 0140 050A 050A 281808 55"}}
{"RfRaw":{"Data":"AA B1 05 0168 0ADC 04D8 00FA 28B4 481828383828382838282808283838282808382828083828382838282808280838283828280828083828280828082838382828382838382828382808382838282838 55"}}
{"RfRaw":{"Data":"AA B1 03 0140 050A 04F6 281808 55"}}
{"RfRaw":{"Data":"AA B1 03 0104 055A 053C 280818 55"}}
{"RfRaw":{"Data":"AA B1 03 014A 050A 0500 281808 55"}}
{"RfRaw":{"Data":"AA B1 03 0140 051E 0514 280818 55"}}
{"RfRaw":{"Data":"AA B1 03 0140 0500 04F6 281808 55"}}
{"RfRaw":{"Data":"AA B1 04 015E 00FA 0514 050A 381828 55"}}
{"RfRaw":{"Data":"AA B1 03 010E 055A 053C 280818 55"}}
{"RfRaw":{"Data":"AA B1 03 0140 0500 0500 281808 55"}}
{"RfRaw":{"Data":"AA B1 03 014A 0528 051E 280818 55"}}
{"RfRaw":{"Data":"AA B1 03 0118 0550 0532 280818 55"}}
{"RfRaw":{"Data":"AA B1 03 0136 051E 0514 281808 55"}}
{"RfRaw":{"Data":"AA B1 03 014A 051E 04D8 280818 55"}}
{"RfRaw":{"Data":"AA B1 03 0140 0514 050A 281808 55"}}
{"RfRaw":{"Data":"AA B1 03 0154 0514 0514 280818 55"}}
{"RfRaw":{"Data":"AA B1 04 015E 00FA 051E 0500 381828 55"}}
``` 

So I conclude that:
a) I need to have raw mode on all the time because otherwise I don't get the signal from the wall switches
b) the remote control doesnt care since its sending commands only, so thats fine
c) the door/window contacts now send different messages via MQTT, and the B1 codes are different each time, so I have no clue on how to hook on those when evaluating MQTT messages
d) same issue as c) for the wall switches

**So**, I don't understand how I should use the wall switches and door contacts in raw mode, which are sending different codes on each press. Of course I could do the B1->B0 conversion, however I don't want to mimic the switches, but instead react on MQTT messages when the switch is pressed. So I would need tasmota/portisch to send a stable code for the wall switch and door/window contacts. How can I do this?

peteakalad, 2021-01-26:
If you have a mix of devices you want to receive then you’ll have to have rfraw on all the time and work out if there’s something in the B1 codes which you can use to identify the device. 

Alternatively maybe look into pushing the rfraw codes through rtl_433 which may support your devices - if it can decode it it can publish to mqtt too.

peteakalad, 2021-01-26:
Actually that rtl_433 change to parse rfraw is not merged but could still try the PR branch.

more info

https://github.com/merbanan/rtl_433/pull/1491

patrickjane, 2021-01-28:
Thanks, will try out

patrickjane, 2021-02-07:
I have yet another issue. While it worked initially when sending B1 codes to the rollerblind, I seem to have a problem with all day regular use. I have set up automations which open the cover every morning, and close it every night.

However, it ocasionally _does not_ work (either it wont go up in the morning, or not go down in the night). To make sure this is not caused by my home automation platform, I have recorded the MQTT messages which are beeing sent. For the last 2 days this looks like:

```
Feb 05 20:45:00 cmnd/tasmota_DB5FEF/backlog rfraw AAB04C040413B002760118139C938192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A19292A1A192A1A1A192A155 ; rfraw 0
Feb 06 07:30:00 cmnd/tasmota_DB5FEF/backlog rfraw AAB04C040413B00280011813BA938192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A19292929292A192A192A155 ; rfraw 0
Feb 06 20:45:00 cmnd/tasmota_DB5FEF/backlog rfraw AAB04C040413B002760118139C938192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A19292A1A192A1A1A192A155 ; rfraw 0
Feb 06 21:34:09 cmnd/tasmota_DB5FEF/backlog rfraw AAB04C040413B002760118139C938192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A19292A1A192A1A1A192A155 ; rfraw 0
Feb 07 07:30:00 cmnd/tasmota_DB5FEF/backlog rfraw AAB04C040413B00280011813BA938192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A19292929292A192A192A155 ; rfraw 0```

So, from the home automation side everything is fine - the MQTT commands are beeing sent.
Yesterday, for example, the rollerblind did not move, although the MQTT message had been sent. Then I manually triggered it again at 21:34, and the cover instantly moved.

So it seems like either tasmota is not sending the signal, or the rollerblind does not catch/accept it ocasionally.

Any hint on how this can be fixed?

nagyrobi, 2021-03-02:
It can be caused by interference. You should re-locate your Tasmota in-house somewhere else (other room).

patrickjane, 2021-03-03:
I have moved the rf bridge to the same room as the motor, at approx 3m straight distance with no obstacles in the way. Still I have the same issues.

To me it rather seems like the converted B1 codes are not accepted all the time. Could this be? E.g. more often the roller blind fails to go up in the morning, but less often the roller blind fails to go down in the evening.

I have generated additional codes and tried different ones, and it seems to depend on the B1 code I am using, whether or not, or how often the command will actually lead to a motor movement. 

So to me it looks like the B0/B1 conversion I have used generated unreliable codes, or I am unable to pick/select the "correct" codes, if such thing exists. However I am still pretty unfamiliar with how this RF protocol works, so basically I have no clue what I am doing beyond what was written in this topic and [here](https://tasmota.github.io/docs/devices/Sonoff-RF-Bridge-433/).

peteakalad, 2021-03-03:
Sometimes the receiver expects the code to be repeated a few times as they go into a deep sleep and need repeated transmissions to waken them up. 

Try running the same rfraw command to transmit the code two or more times back to back to see if it helps.

Also try holding the original remote button down for longer when capturing the B1 code, it may copy repeated signal do you don’t need to do the above?

patrickjane, 2021-03-03:
Okay so you mean instead of:

```
Topic=cmnd/sonoff-rf-bridge/backlog
Payload=rfraw AA B0 4C 04 04 13B0 0276 0118 139C 938192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A19292A1A192A1A1A192A1 55; rfraw 0
```

I would be doing something like:

```
Topic=cmnd/sonoff-rf-bridge/backlog
Payload=rfraw AA B0 4C 04 04 13B0 0276 0118 139C 938192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A19292A1A192A1A1A192A1 55; rfraw AA B0 4C 04 04 13B0 0276 0118 139C 938192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A19292A1A192A1A1A192A1 55; rfraw AA B0 4C 04 04 13B0 0276 0118 139C 938192A192A1A1A19292A192A1A1A192A1929292A192A192A1929292A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A19292A1A192A1A1A192A1 55; rfraw 0
```

Or should I send several MQTT messages?

peteakalad, 2021-03-03:
Yes something like that see if it helps make the blind more responsive.  Also can try with a short delay with the delay command in between rfraw b0 commands e.g. Delay 2 would pause for 200ms between transmissions, also might help.

patrickjane, 2021-03-03:
Alright, thanks for the support, will try out.

patrickjane, 2021-03-15:
Thanks, seems to have done the trick.

closed 2021-03-15 by patrickjane

nagyrobi, 2021-03-15:
I had the same problem with covers which had to open simultaneously. Often only first cover moved. It was likely caused by commands sent too fast and switching back and forth the RfRaw sniffing with values and 0.

I think backlogs with `rfraw AAB0XXXXX....XXXXXXXXXX; rfraw 0` should be avoided as switching the sniff on and off quickly for every cover movement may cause issues. It's enough to send rfraw 0 only once from HA system with some delay after all procedures related to cover movements finished. No backlog needed...