Portisch/RF-Bridge-EFM8BB1 #193: Upload error code 209/211/212

RolandFran, 2021-03-11:
Hi

I tried to upgrade the RF-firmware on a R2 revision of the Sonoff RF Bridge that I got today
with portisch "RF-Bridge-EFM8BB1-20190220.hex".
After about 5 to 10 seconds the upload failed with "Upload error code 209" and later 211 and also 212

---------------------
Console:
00:00:00.049 CFG: Loaded from flash at FB, Count 35
00:00:00.054 SER: Set to 8N1 19200 bit/s
00:00:00.055 QPC: Count 1
00:00:00.060 SER: Set to 8N1 19200 bit/s
00:00:00.063 Project tasmota Tasmota Version 9.3.1(tasmota)-2_7_4_9(2021-03-09T16:12:28)
...
21:56:17.043 UPL: File portisch506.hex
21:56:18.618 UPP: Multicast (re)joined
21:56:18.619 UPL: Transfer 22702 bytes
--------------------
I tried already other jumper cables, cables other way around and switch on instead off -> same result
Any idea what that could be?

ricardo-cruz44, 2021-03-11:
What card are you using to flash the EFM8Bb?
I have a Sonoff RF. Bridge Rev 2 that I flashed last week no problem.

RolandFran, 2021-03-12:
yes, that's the same one I have.
Flashing Tasmota was no problem, but the portisch firmware update failed.

ricardo-cruz44, 2021-03-12:
Sorry for the confusion, on the RF bridge rev2 for flashing the EFM8BB I used an Arduino uno card to control the C2d and C2ck pins and upload the hex file to the EFM8BB, I have not done the Tasmota flash upgrade yet, still deciphering the  capture data.

ElwynnEldoriath, 2021-04-07:
> yes, that's the same one I have.
> Flashing Tasmota was no problem, but the portisch firmware update failed.

Have you been able to solve the issue? I am having the same errors (code 211) when trying to upload the Portisch firmware.

RolandFran, 2021-04-08:
Unfortunately not, spent several hours trying to get the bridge working, till now without any success

Mindsuk, 2021-04-09:
try to flash Tasmota again and than Portisch again

ElwynnEldoriath, 2021-04-12:
> try to flash Tasmota again and than Portisch again

Did it multiple times, no change... always getting error 211

waynen1967, 2021-04-17:
Total newbie here but I followed this youtube video and had no issues installing Tasmota and Portisch

ElwynnEldoriath, 2021-04-19:
Watched it before, no difference...

cph25, 2021-04-19:
I was running in the same issue and very close to give up. The video by DrZzs shows a successful flashing of the RF chip. There was one difference, he used the Tasmota 8.4.0 and the English version, downloaded the 8.4.0  from https://github.com/arendst/Tasmota/releases/, flashed the bridge via Tasmotizer and it worked.

ElwynnEldoriath, 2021-04-20:
Still getting a "Failed to init RF chip". Either the device is broken or there is a different hardware rev. circling

cph25, 2021-04-20:
I got this too multiple times: check if the 2 wires are well connected, play with the switch s2 - I'm no more sure when it worked.

ElwynnEldoriath, 2021-04-20:
I checked the proper connection with a multimeter, tried both switch position: no change

chrisvor, 2021-05-29:
Just a short note that I was having the exact same issue with Tasmota 9.4.0. Then I saw the post by cph25 mentioning that DrZzs was using Tasmota 8.4.0..... So I thought to give it a try... Reflashed 8.4.0 -> Firmware upgrade -> selected the portisch 2019 hex file -> boom! Flash successful!
I then updated tasmota to 9.4.0 (minimal first, then full version) and I am now on 9.4.0 with portisch firmware working!

ktuulos, 2021-08-23:
On my case, this worked fine with Tasmota 9.5.0 and RF-Bridge-EFM8BB1-20190220.hex. RF bridge was powered externally with 3.3 volts and GPIO lines were not cut away. There were just two flashing wires between MCU and EFM8BB1.

kiat-huang, 2021-09-18:
> On my case, this worked fine with Tasmota 9.5.0 and RF-Bridge-EFM8BB1-20190220.hex. RF bridge was powered externally with 3.3 volts and GPIO lines were not cut away. There were just two flashing wires between MCU and EFM8BB1.

Same experience here, except I had the two wires attached as per the instructions...

[image]

Robmak73, 2021-09-18:
I had success with same setup. The wire cut is not needed if you power the
chip directly if I remember right.
I did brick my first bridge but put this down to the ft232 ttl adapter.
Managed to tamotize straight away cp2102

arbasel, 2021-09-25:
I have the Sonoff RF Bridge R2 but v2
[image]
@Robmak73  could you please assist me with which pins I need to connect to where? From what you say I should not need to cut any wires.

jgruiz68, 2021-10-06:
@arbasel Did you have any luck with the board? I have the same version.

Robmak73, 2021-10-06:
Yes I used the tasmotizer and the software did everything for me. 
No need to cut wires on the board. Just make sure you father power up procedure and connect the 2 sets of 2 pins together. I did brick a sonoff bridge but found that it was due to the usb ttl converter from ebay was the wrong type. Soon as I got the right usb to ttl adapter I had no problem at all.

jgruiz68, 2021-10-06:
I used tasmotizer to flash Tasmota as well, but did you also flash the portisch firmware using tasmotizer? The part I am having trouble with is flashing the portisch firmware.
> 
> 
> Yes I used the tasmotizer and the software did everything for me. No need to cut wires on the board. Just make sure you follow the power up procedure and connect the 2 sets of 2 pins together. I did brick a sonoff bridge but found that it was due to the usb ttl converter from ebay was the wrong type. Soon as I got the right usb to ttl adapter I had no problem at all.

joncknott, 2021-10-06:
> I used tasmotizer to flash Tasmota as well, but did you also flash the portisch firmware using tasmotizer? The part I am having trouble with is flashing the portisch firmware.
> 

@jgruiz68 I'm having the same issue unfortunately. From a [message in the Discord channel about Tasmota Flashing](https://discord.com/channels/479389167382691863/790182241908686870/892472267764805664), it seems that the R2 V2.2 boards have a different RF chip:

> As far as I can tell, the old EFM8BB1 has been replaced by an 8R08A

So it looks like this firmware will not work on this version of the Sonoff Bridge.

jgruiz68, 2021-10-06:
@joncknott Thanks for the reference, I will keep an eye on new developments. In the mean time I will look for an older bridge.

Robmak73, 2021-10-06:
No I did not use tasmotizer as it does not do the Portishead flash. Its relatively simple if you get the power correct. The wifi does not have to be on and the hex file is small. I used node mcu as per wiki information. It is really easy if you follow the process exactly Sent from Samsung Mobile on O2

Robmak73, 2021-10-06:
I don't think that is necessary. Just make sure you power the chip directly with 3.3v when flashing rf433 chip. If you look at the circuit board you can see the pin holes for independent power. The wiki explains it much better than me. But the board only needs the 2 conducting strips cutting if you trying to power the wifi chip and rf433 chip on the same adapter. The wifi chip does not need power to flash the rf433 chip. So don't cut the board. Just move your + and - from the wifi input to the rf433 input. The board is labelled if you look closely. To double check follow the track to show the exact same result by using this method to cutting the board. The rf433 chip is much easier to flash. Infact it was done so quickly I thought that it had not worked. The hex (if I remember correctly) file is relatively small. Sent from Samsung Mobile on O2

TomLyon, 2021-11-21:
The older model rf bridge is disappearing from online stores. I've got 2 of the newer r2 v2 models and would love to flash this firmware if anybody has figured it out yet.

grf21a, 2022-04-09:
Hi,  Hope this may help some people experiencing the 209 error.

I flashed my Black RF Bridge today with Tasmota 11.0.0 and then proceeded to install Portisch following the video available by DrZzs.

During the installation of Portisch firmware I was getting the "Upload error code 209" message appearing and went searching for an answer but could not find anything that would help.

I checked that the filename had the correct file extension as mentioned by DrZzs.  For those that do not know already, it is important that within Win10 File Explorer, click on the View menu, then select the tick box in "File Name Extensions" (see attached graphic). You can also go to a windows command prompt and navigate to the folder that contains the file and run the DIR command to double check.

The file name was not the root cause in my case so I opened up the file in Notepad to double check that I has pasted contents correctly and again everything looked okay initially.  I scrolled to the bottom of the file and noticed that cursor was positioned at the end of the last line, so I decided to hit enter on this line to move the cursor to the next line so the file would end on a blank line (see attached)

When I ran the Firmware Upgrade again everything worked successfully, so it ended up being a simple carriage return/line feed that caused my issue.

Anyway, hope this helps other people out there.  Good luck
[image]
[image]
.