Portisch/RF-Bridge-EFM8BB1 #194: Help to create new protocol

patrickjane, 2021-03-20:
Hello,

I have bought several [wall switches](https://www.trust.com/en/smarthome), and I would like to integrate them into my smart home. 
I have already tried sniffing the B1 codes using `rfraw 177`, and I will get something like:

```
11:10:43 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:10:43","RfRaw":{"Data":"AA B1 05 0168 0AB4 04C4 00E6 280A 481828383828382828383828382838283828283838283828283838283828283828083828382838282838283838282808280828382838382838282838382838282838 55"}}
11:10:43 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:10:43","RfRaw":{"Data":"AA B1 04 014A 00F0 04F6 04EC 381828 55"}}
11:10:43 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:10:43","RfRaw":{"Data":"AA B1 05 015E 0AAA 04C4 00E6 2800 481828383828382828383828382838283828283838283828280838283828280828083828382838282808280838282808280828082808382838282808382838282838 55"}}
11:10:43 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:10:43","RfRaw":{"Data":"AA B1 03 014A 04F6 04EC 280818 55"}}
11:10:56 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:10:56","RfRaw":{"Data":"AA B1 03 014A 04EC 04EC 281808 55"}}
11:10:56 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:10:56","RfRaw":{"Data":"AA B1 05 015E 0AAA 04C4 00E6 280A 481828383828382828383828382838283828280838283828283838283828280828083828382838282808280838282808280828082808382838282808382838282838 55"}}
11:10:56 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:10:56","RfRaw":{"Data":"AA B1 03 0140 04F6 04E2 280818 55"}}
11:10:56 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:10:56","RfRaw":{"Data":"AA B1 05 015E 0AB4 04C4 00E6 280A 481828383828382828383828382838283828280838283828280838283828280828083828382838282808280838282808280828082808382838282808382838282838 55"}}
11:10:56 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:10:56","RfRaw":{"Data":"AA B1 04 014A 00F0 04F6 04EC 381828 55"}}
11:11:05 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:05","RfRaw":{"Data":"AA B1 03 0140 0500 04E2 280818 55"}}
11:11:05 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:05","RfRaw":{"Data":"AA B1 05 014A 0AD2 04D8 00F0 2828 481828083828382828080828082808280828280808280828280808280828280828080828082808282808280808282808280828082808082808282808082808282808 55"}}
11:11:05 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:05","RfRaw":{"Data":"AA B1 03 014A 0500 04EC 280818 55"}}
11:11:05 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:05","RfRaw":{"Data":"AA B1 05 015E 0AB4 04C4 00E6 280A 481828383828382828383828382838283828280838283828280838283828280828083828382838282808280838282808280828082808382838282808382838282838 55"}}
11:11:05 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:05","RfRaw":{"Data":"AA B1 04 014A 00F0 04F6 04EC 381828 55"}}
11:11:05 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:05","RfRaw":{"Data":"AA B1 05 0168 0AB4 04C4 00E6 280A 481828383828382828383828382838283828283838283828283838283828283828083828382838282838283838282838280828382838382838282808382838282838 55"}}
11:11:06 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:06","RfRaw":{"Data":"AA B1 03 0140 0500 04EC 280818 55"}}
11:11:07 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:07","RfRaw":{"Data":"AA B1 03 014A 04F6 04D8 280818 55"}}
11:11:08 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:08","RfRaw":{"Data":"AA B1 04 0140 0AD2 04E2 2814 381828080828082828080828082808280828280808280828280808280828280828080828082808282808280808282808280828082808082808282808082808282808 55"}}
11:11:08 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:08","RfRaw":{"Data":"AA B1 03 014A 04E2 04E2 281808 55"}}
11:11:08 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:08","RfRaw":{"Data":"AA B1 05 015E 0AAA 04CE 00E6 27F6 481828383828382828383828382838283828280838283828280838283828280828083828382838282838280838282808280828082808382838282808382838282838 55"}}
11:11:08 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:08","RfRaw":{"Data":"AA B1 04 014A 00F0 04F6 04E2 381828 55"}}
11:11:08 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:08","RfRaw":{"Data":"AA B1 05 0168 0AB4 04C4 00E6 2800 481828383828382828383828382838283828283838283828283838283828283828083828382838282808283838282838280828082838382838282838382838282838 55"}}
11:11:08 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:08","RfRaw":{"Data":"AA B1 04 014A 00F0 04F6 04E2 381828 55"}}
11:11:09 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:09","RfRaw":{"Data":"AA B1 04 0172 00D2 04D8 047E 381828 55"}}
11:11:09 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:09","RfRaw":{"Data":"AA B1 03 0140 04F6 04F6 281808 55"}}
11:11:09 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:09","RfRaw":{"Data":"AA B1 05 014A 0AC8 04E2 00F0 2828 481828083828082828080828082808280828280808280828280808280828280828080828082808282808280808282808280828082808082808282808082808282808 55"}}
11:11:09 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:09","RfRaw":{"Data":"AA B1 03 014A 0500 04EC 280818 55"}}
11:11:10 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:10","RfRaw":{"Data":"AA B1 05 015E 0AB4 04C4 00E6 2800 481828383828382828383828382838283828280838283828283838283828280828083828382838282808280838282808280828082808382838282808382838282838 55"}}
11:11:10 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:10","RfRaw":{"Data":"AA B1 03 0140 0500 04EC 280818 55"}}
11:11:10 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:10","RfRaw":{"Data":"AA B1 05 015E 0AB4 04C4 00F0 2814 481828383828382828383828382838283828280838283828280838283828280828083828382838282808280838282808280828082808382838282808382838282838 55"}}
11:11:10 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:10","RfRaw":{"Data":"AA B1 03 014A 04F6 04EC 280818 55"}}
11:11:11 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:11","RfRaw":{"Data":"AA B1 03 014A 0500 04E2 280818 55"}}
11:11:12 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:12","RfRaw":{"Data":"AA B1 05 014A 0AD2 04E2 00F0 2832 481828083828082828080828082808280828280808280828280808280828280828080828082808282808280808282808280828082808082808282808082808282808 55"}}
11:11:12 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:12","RfRaw":{"Data":"AA B1 03 014A 0500 04EC 280818 55"}}
11:11:12 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:12","RfRaw":{"Data":"AA B1 05 015E 0AAA 04CE 00E6 2800 481828383828382828383828382838283828280838283828280838283828280828083828382838282808280838282808280828082808382838282808382838282838 55"}}
11:11:12 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:12","RfRaw":{"Data":"AA B1 03 0140 0500 04E2 280818 55"}}
11:11:12 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:12","RfRaw":{"Data":"AA B1 05 0168 0AB4 04C4 00F0 2800 481828383828382828383828382838283828283838283828283838283828283828083828382838282838283838282808280828082838382838282808382838282838 55"}}
11:11:12 MQT: tele/tasmota_DB5FEF/RESULT = {"Time":"2021-03-20T11:11:12","RfRaw":{"Data":"AA B1 04 014A 00F0 04F6 04E2 381828 55"}}
```

Now, I saw that it is possible to [define new protocols](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/Define-new-protocol-in-RF_Protocols.h), so I thought maybe I can try this.

Is my understanding correct, that, when properly added, a new protocol leads to the tasmota software beeing able to spit out "normal" codes/payloads without using the `rfraw` modes, e.g. like `{"Sync":14110,"Low":420,"High":1360,"Data":"E2BC0E","RfKey":"None"}`?

If so, I would have several questions, since the documentation for adding new protocols is not 100% clear to me, and I am not familiar with RF protocols.

1) What does *The longest bucket found is taken as sync bucket and is placed as first bucket number in the sniffed data: ->3<- 818190908181908190909090908190819081818190909091A* mean? My understanding of "bucket" is "block of bytes", and in e.g. `AA B1 05 015E 0AB4 04C4 00F0 2814 481828383828382828383828382838283828280838283828280838283828280828083828382838282808280838282808280828082808382838282808382838282838 55` the "longest bucket" is the one starting with `48182...`. So this whole block should be the "sync bucket"? Then what does "is placed as first bucket number in the sniffed data" mean? Does it mean I should  [... cut ...]
13) In the end, it seems like the numbers from 11) shall be converted using `& 0x77` with each "number", e.g. this would equal:
```
8182808382808282808082808280828082828080828082828080828082828082808082808280828280828080828280828082808280808280828280808280828280
1010001010001010000010001000100010100000100010100000100010100010000010001000101000100000101000100010001000001000101000001000101000
569aa596
```
 = 32 bit
Is that correct!?

14) I dont understand how to evaluate BIT0 and BIT1
15) How do I know whether or not there is a pause between transmit repeats?

If someone could help me out I would be happy :)

Robmak73, 2021-10-06:
Any progress with your problem. I have exactly the same problem with different rf sensors. Have you made any progress yourself. The information on this is very poor for such a complicated process which requires a understanding of lots of different subjects. Most are things that we don't normally have any need to get acquainted with.

kajarit, 2022-09-25:
Hi @patrickjane,

I'm also trying to decode some 433 sensor data, and have similar issues understanding the wiki guide, especially "buckets" and "nibbles".
My captured data also contains similar RF data part like yours, but when I try to remove the "high/low bucket marking by doing a logical and by 0x77 with each data byte" I get non-binary result:
`81828281828282828282818282828282828282828181818181818181828282828282828283`
after conversion, I get which is definitely non-binary:
`01020201020202020202010202020202020202020101010101010101020202020202020203`

How did you do the & 0x77 operation exactly?

> 13. In the end, it seems like the numbers from 11) shall be converted using `& 0x77` with each "number", e.g. this would equal:
>
> ``` [...]