Portisch/RF-Bridge-EFM8BB1 #195: Adding a new protocol to the firmware

StefanM99, 2021-03-22:
Hi,

I want to add a new protocol for Retekess pagers. 

Here is the sniffed data:

{"RfRaw":{"Data":"AA B1 04 0118 0230 0096 163A 3818181929292928181818181818181818192928181819281818 55"}}

I added to RF-Protocols.h:

		/*
		 * Retekess TD158
		 */
		{
			{ &PROTOCOL_PULSES(TD158), ARRAY_LENGTH(PROTOCOL_PULSES(TD158)) },
			{ &PROTOCOL_START(TD158), ARRAY_LENGTH(PROTOCOL_START(TD158)) },
			{ &PROTOCOL_BIT0(TD158), ARRAY_LENGTH(PROTOCOL_BIT0(TD158)) },
			{ &PROTOCOL_BIT1(TD158), ARRAY_LENGTH(PROTOCOL_BIT1(TD158)) },
			25,
			false

..now I try to compile a new firmware file and I do not really find any instructions on howto do that.

I downloaded Simplicity Studio and then I am stuck...

Can anybody help?

Thanks

StefanM99, 2021-03-23:
Solved the problem

closed 2021-03-23 by StefanM99

Arganox, 2021-12-28:
@StefanM99 sorry for reviving this issue. But the rest of the project seems rather dead :(

Could you eleborate a little more on this? what did you do to install SS4 and with what settings?
Also, could you maybe help me figure out the BIT0 and BIT1 values? Reading the doc I'm not sure how to determine these values.

(btw, I've opened an issue for this as well.: #212  )

Arganox, 2021-12-28:
@StefanM99 Got the Simplicity studio running. Tried it in a Windows VM this was pretty straight forward. Failed on Debian though :(

That leaves just the BIT0 and BIT1 value interpretation. If you would be so kind to help me out here?
I'll be doing some trial-and-error in the meanwhile I guess.