Portisch/RF-Bridge-EFM8BB1 #197: Dooya DC90 remote, not working with RF bridge

vramich, 2021-03-26:
Sonoff RF Bridge V2 flashed with Tasmota v9.3.1 Kenneth
EFM8BB1 flashed with the latest version of Portisch Version 0x04
### Information

Dooya DC90 remote
https://photos.app.goo.gl/Nps4fAgA2LCqzsX79 
https://www.four-more.net/haus-garten/garten-sonnenschutz/markisen/ersatzfernbedienung-fuer-markise/a-60940?ReferrerID=7&gclid=Cj0KCQjwjPaCBhDkARIsAISZN7RkX0YI88pQLMCjUFY9Hmu8sZsl_H9JmNQK9IDVm6CRBeBnBU7NuuEaArRvEALw_wcB

### Sniffed data (rfraw 0xB1)
**OPEN**
**long press**
12:43:06 RSL: tele/tasmota_C244C3/RESULT = {"Time":"2021-03-26T12:43:06","RfRaw":{"Data":"AA B1 05 12C0 05E6 015E 02C6 1E32 481A3B2A3B2A3A3B2A3A3A3A3A3B2B2A3A3B2B2B2A3A3A3B2B2A3B2A3B2A3A3A3B2A3A3A3B2A3A3A3B281A3B2A3B2A3A3B2A3A3A3A3A3B2B2A3A3B2B2B2A3A3A3B2B2A3B2A3B2A3A3A3B2A3A3A3B2A3A3A3B 55"}}

"RfRaw":{"Data":"AA B1 05 12B6 05F0 0154 02D0 1E82 481A3B2A3B2A3A3B2A3A3A3A3A3B2B2A3A3B2B2B2A3A3A3B2B2A3B2A3B2A3A3A3B2A3A3A3B2A3A3A3B281A3B2A3B2A3A3B2A3A3A3A3A3B2B2A3A3B2B2B2A3A3A3B2B2A3B2A3B2A3A3A3B2A3A3A3B2A3A3A3B 55"}}

"RfRaw":{"Data":"AA B1 05 12C0 05E6 015E 02C6 1E32 481A3B2A3B2A3A3B2A3A3A3A3A3B2B2A3A3B2B2B2A3A3A3B2B2A3B2A3B2A3A3A3B2A3A3A3B2A3A3A3B281A3B2A3B2A3A3B2A3A3A3A3A3B2B2A3A3B2B2B2A3A3A3B2B2A3B2A3B2A3A3A3B2A3A3A3B2A3A3A3B 55"}}

**short**
"RfRaw":{"Data":"AA B1 05 0604 014A 02D0 1E32 12C0 C092A192A19292A19292929292A1A19292A1A1A1929292A1A192A192A1929292A1929292A1929292A3 55"}}

"RfRaw":{"Data":"AA B1 05 05E6 0154 02D0 1E28 12B6 C092A192A19292A19292929292A1A19292A1A1A1929292A1A192A192A1929292A1929292A1929292A3 55"}}

**STOP**
12:43:33 RSL: tele/tasmota_C244C3/RESULT = {"Time":"2021-03-26T12:43:33","RfRaw":{"Data":"AA B1 05 05FA 0154 02D0 1F72 12A2 C092A192A19292A19292929292A1A19292A1A1A1929292A1A192A192A1929292A1929292A1A1A1A193 55"}}

"RfRaw":{"Data":"AA B1 05 05F0 014A 02DA 1E28 12B6 C092A192A19292A19292929292A1A19292A1A1A1929292A1A192A192A1929292A19292A1A19292A1A3 55"}}

"RfRaw":{"Data":"AA B1 05 05F0 014A 02DA 1E32 12B6 C092A192A19292A19292929292A1A19292A1A1A1929292A1A192A192A1929292A19292A1A19292A1A3 55"}}

**CLOSE** 
12:46:10 RSL: tele/tasmota_C244C3/RESULT = {"Time":"2021-03-26T12:46:10","RfRaw":{"Data":"AA B1 05 12B6 05F0 0154 02D0 1EA0 481A3B2A3B2A3A3B2A3A3A3A3A3B2B2A3A3B2B2B2A3A3A3B2B2A3B2A3B2A3A3A3B2A3A3B2B2A3A3B2B281A3B2A3B2A3A3B2A3A3A3A3A3B2B2A3A3B2B2B2A3A3A3B2B2A3B2A3B2A3A3A3B2A3A3B2B2A3A3B2B 55"}}

"RfRaw":{"Data":"AA B1 05 12AC 05E6 015E 02D0 1E3C 481A3B2A3B2A3A3B2A3A3A3A3A3B2B2A3A3B2B2B2A3A3A3B2B2A3B2A3B2A3A3A3B2A3A3B2B2A3A3B2B281A3B2A3B2A3A3B2A3A3A3A3A3B2B2A3A3B2B2B2A3A3A3B2B2A3B2A3B2A3A3A3B2A3A3B2B2A3A3B2B 55"}}

"RfRaw":{"Data":"AA B1 05 12B6 05E6 014A 02D0 1E32 481A3B2A3B2A3A3B2A3A3A3A3A3B2B2A3A3B2B2B2A3A3A3B2B2A3B2A3B2A3A3A3B2A3A3B2B2A3A3B2B281A3B2A3B2A3A3B2A3A3A3A3A3B2B2A3A3B2B2B2A3A3A3B2B2A3B2A3B2A3A3A3B2A3A3B2B2A3A3B2B 55"}}

###  rfraw 177
**OPEN**
"RfRaw":{"Data":"AA B1 05 12B6 05E6 0154 02DA 1E8C 481A3B2A3B2A3A3B2A3A3A3A3A3B2B2A3A3B2B2B2A3A3A3B2B2A3B2A3B2A3A3A3B2A3A3A3B2A3A3A3B281A3B2A3B2A3A3B2A3A3A3A3A3B2B2A3A3B2B2B2A3A3A3B2B2A3B2A3B2A3A3A3B2A3A3A3B2A3A3A3B 55"}}

**STOP** 

"RfRaw":{"Data":"AA B1 05 05FA 0154 02DA 1E28 12B6 C092A192A19292A19292929292A1A19292A1A1A1929292A1A192A192A1929292A1929292A1929292A3 55"}}

**CLOSE** 
13:08:34 RSL: tele/tasmota_C244C3/RESULT = {"Time":"2021-03-26T13:08:34","RfRaw":{"Data":"AA B1 05 12B6 05F0 0154 02D0 1EAA 481A3B2A3B2A3A3B2A3A3A3A3A3B2B2A3A3B2B2B2A3A3A3B2B2A3B2A3B2A3A3A3B2A3A3B2B2A3A3B2B281A3B2A3B2A3A3B2A3A3A3A3A3B2B2A3A3B2B2B2A3A3A3B2B2A3B2A3B2A3A3A3B2A3A3B2B2A3A3B2B 55"}}
13:08:34 RSL: tele/tasmota_C244C3/RESULT = {"Time":"2021-03-26T13:08:34","RfRaw":{"Data":"AA B1 05 05FA 014A 02DA 1E3C 12C0 C092A192A19292A19292929292A1A19292A1A1A1929292A1A192A192A1929292A19292A1A19292A1A3 55"}}

**Some times I recieve two codes for 1 push**
**close** 
13:08:34 RSL: tele/tasmota_C244C3/RESULT = {"Time":"2021-03-26T13:08:34","RfRaw":{"Data":"AA B1 05 12B6 05F0 0154 02D0 1EAA 481A3B2A3B2A3A3B2A3A3A3A3A3B2B2A3A3B2B2B2A3A3A3B2B2A3B2A3B2A3A3A3B2A3A3B2B2A3A3B2B281A3B2A3B2A3A3B2A3A3A3A3A3B2B2A3A3B2B2B2A3A3A3B2B2A3B2A3B2A3A3A3B2A3A3B2B2A3A3B2B 55"}}
13:08:34 RSL: tele/tasmota_C244C3/RESULT = {"Time":"2021-03-26T13:08:34","RfRaw":{"Data":"AA B1 05 05FA 014A 02DA 1E3C 12C0 C092A192A19292A19292929292A1A19292A1A1A1929292A1A192A192A1929292A19292A1A19292A1A3 55"}}

The command always gets acknowledged (AAA055 code), the red LED on the sonoff bridge blinks (to show RF data was transmitted). But nothing happens. 

Anybody got any ideas?? Please help

runeka, 2021-04-07:
I am stuck at the same point you are. I also tried the Esphome/Portisch route. Still stuck at nothing happens.

RasputinBerry, 2021-04-26:
Try this ...
#202

But I have (also) problems with the SONOFF ioBroker adapter. I can't get the right MQTT settings here. RfRaw Data is not filled. And I don't know where in the objects I have to enter the send command.

vramich, 2021-04-26:
@RasputinBerry

It works, It works !!!!

Thank you!!!!

for me:
**OPEN**
rfraw AAA80601520CE3511155

**STOP**
rfraw AAA8060B520CE3515555

**CLOSE**
rfraw AAA80601520CE3513355

closed 2021-04-26 by vramich

RasputinBerry, 2021-04-26:
OK, I'm glad it works. How do you send the codes? With MQTT? Any idea for "my" problem?

vramich, 2021-04-26:
@RasputinBerry 
console at the moment. I will try with MQTT, and come back to you.

RasputinBerry, 2021-04-26:
The Tasmota console is also my solution, but it would be nice if you can get MQTT to work. Looking forward to hear from you.

vramich, 2021-04-27:
@RasputinBerry 
works for me:

cover:
  - platform: mqtt
    name: "shutter1"
    command_topic: "cmnd/tasmota_C244C3/backlog"
    qos: 1
    payload_close: "rfraw AAA80601520CE3513355; RfRaw 0"
    payload_stop: "rfraw AAA8060B520CE3515555; RfRaw 0"
    payload_open: "rfraw AAA80601520CE3511155; RfRaw 0"
    optimistic: true

RasputinBerry, 2021-04-27:
OK, but that's not the SONOFF adapter in the IoBroker, right? Unfortunately, these settings do not work for me.