Portisch/RF-Bridge-EFM8BB1 #198: Error Code 211 while upgrading to Portisch

ElwynnEldoriath, 2021-04-07:
Since there are others with the issue and the discord is not able to help, I am trying here again:

I'm using a Sonoff V2.0 RF bridge that has been successfully upgraded to Tasmota, set to device 25 (RF bridge), usb connectors on the PCB have been scratched, cables are soldered properly (3.3V, Gnd; TX & RX for the tasmota flash; the two cables during Portisch flash soldered as well with no crossing).
Switch position OFF and ON have been tested, the hex file(s) tested are correct, microUSB as well as TTL-powering have been used etc. 
I'm getting error code 211 every time. 

Any advice? :(

Ierlandfan, 2021-06-19:
For future Reference, I stumbled  into the same issue. Error 210 and 211. Use Tasmota sensors binary. Do not use the default tasmota.bin

whi4, 2021-06-23:
Hi,

I found this link useful:

programming notes for tasmota:
https://tasmota.github.io/docs/devices/Sonoff-RF-Bridge-433/

I hope this helps you, but I found that the Portisch software does not work with the bucket sniffing (raw).

The latest bin version is from 2018, but some issues from 2020 seem to be similar to my problem. There are have been solutions proposed but I do not have the skills to create my own bin.

Good Luck

Regards,

Robert

CMaxim77, 2021-11-29:
hello, sorry to post on a old topic. im trying to update my rf Bridge with firmware Keil 8051 v9.53 - Release
https://github.com/Portisch/RF-Bridge-EFM8BB1/tree/master/Keil%208051%20v9.53%20-%20Release

I keep getting  "Upload error code 209" - Im using 10.0.0(tasmota) I have tried withthe R switch both On and Off.
I have a few basic rf remotes that I can get the codes for but ther are two remotes that don't se the codes at all, from what I have researched this firmware may let me see codes that the origional firmware won't sniff.
If anybody could help or oint me whereI can find some information I would be grateful as I can't find anything on this.

Thank you, Corey
[image]