Portisch/RF-Bridge-EFM8BB1 #199: Smarteres Wallswitch SH5-TSW-A

Mindsuk, 2021-04-09:
### Latest firmware used?
I flashed the Sonoff Bridge with this Hex File:

RF-Bridge-EFM8BB1-20190220.hex

### Information

Smarteres Wallswitch SH5-TSW-A

### Sniffed data **(Button on and off) with RfRaw 166**
12:52:26.037 MQT: tele/rf-bridge/RESULT = {"Time":"2021-04-09T12:52:26","RfRaw":{"Data":"AAA6050D33A5FFED55"}}
12:52:30.634 MQT: tele/rf-bridge/RESULT = {"Time":"2021-04-09T12:52:30","RfRaw":{"Data":"AAA6050D33A5FFFD55"}}

Do I have to convert the Code within the BitBucket Converter to use it within MQTT and Homeassistant? If I tried it, the message is: 

###### Skipping line '{"RfRaw":{"Data":"AAA6050D33A5FFED55"}}' ######

patrickjane, 2021-04-13:
Please let me know when you find a solution :)

patrickjane, 2021-05-29:
Did you ever figure out how to use the switches?

@Portisch Can you help out?