Portisch/RF-Bridge-EFM8BB1 #200: trying to make this make sense, sorry

docsparks, 2021-04-10:
Im sorry to post this, but I feel like my brain is broken and after wasting too much time, I might as well ask for help.

I cant get (windows10) python3 to run the bitbucketconverter.py, it just opens a new window for me, and even then, copying from the tasmota log and the process has been giving me nothing but headaches.  resorted to web interface but regardless, I have written some short visual basic tool for myself, but i still can not figure out how to make this make sense.

https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/178
on that page it refers to 'length' in 'µs' (microsiemen?) and i have no idea where that means, 
and the length (XX) of the AAB1 XX is never the amount that is in the data to transmit, so if anyone could shed some light on how i can convert the math, I would really appreciate it.  I dont know python well, just dabble in as many languages as i have to.  

at the moment i have

in:
AA B1 04 019A 0320 0104 24B8 3819090908190909090818190819090818181819090818181A 55

out:
AA B0 23 04 08 019A 0320 0104 24B8 3819090908190909090818190819090818181819090818181A 55

broken down:
0 ) AA - AA
1 ) B1 - B0
2 ) 04 - 23
2A) ?? - 04
2B)  ?? - 08
3 ) 019A
4 ) 0320
5 ) 0104
6 ) 24B8
7 ) 3819090908190909090818190819090818181819090818181A
8 ) 55

and for the cherry ontop, if anyone is familiar with visual basic / vb.net / visual studio,  I have this breaking down the in/out via spaces..

`
        Dim parts = input_txt.Split(" ")
        Dim cnt = parts.Length.ToString - 1
        Dim line As String     
        Dim LineText = String.Empty

        For StartNum = 0 To cnt
            line = UCase(Trim(parts(StartNum)))
            LineText &= (StartNum & " ) " & line)

            If Not StartNum = cnt Then LineText &= vbNewLine
        Next
        RichTextBox4.Text = LineText
`

no idea how i get the values for 2A, 2B, other than using the online tool.
even finding out how to get github to paste my crummy vb code correctly is killing my alotted time.

please jeebus help.

waynen1967, 2021-04-17:
I'm no help, brand new to all this, ut how did you "break down" the commands like that?

docsparks, 2021-04-29:
> I'm no help, brand new to all this, ut how did you "break down" the commands like that?

This is just separating the command by spaces, and in the visual studio application