Portisch/RF-Bridge-EFM8BB1 #202: DOOYA DC 90 and EMPASA sun protection (tubular motor)

RasputinBerry, 2021-04-24:
### Latest firmware used?
... yes ... TASMOTA 9.3.1 and RF-Bridge-EFM8BB1-20190220,hex

### Information
....is in the subject 

### Sniffed data
I use the "RfRaw 166" and get clean signals ... 

`UP
07:43:48.216 MQT: tele/bridge/RESULT = {"Time":"2021-04-24T07:43:48","RfRaw":{"Data":"AAA606012A1A15D11155"}}
07:43:48.265 MQT: tele/bridge/RESULT = {"Time":"2021-04-24T07:43:48","RfRaw":{"Data":"AAA606012A1A15D11E55"}}

STOP
07:44:22.211 MQT: tele/bridge/RESULT = {"Time":"2021-04-24T07:44:22","RfRaw":{"Data":"AAA6060B2A1A15D15555"}}
07:44:39.579 MQT: tele/bridge/RESULT = {"Time":"2021-04-24T07:44:39","RfRaw":{"Data":"AAA606012A1A15D15555"}}

DOWN
07:45:07.638 MQT: tele/bridge/RESULT = {"Time":"2021-04-24T07:45:07","RfRaw":{"Data":"AAA606012A1A15D13355"}}
07:45:07.736 MQT: tele/bridge/RESULT = {"Time":"2021-04-24T07:45:07","RfRaw":{"Data":"AAA606012A1A15D13C55"}}
`
My problem is that the command
"cmnd / bridge / Backlog RfRaw AAA606012A1A15D11E55; RfRaw 0"
doesn't work, the motor doesn't move ...
The problem is sitting in front of the computer again, right? What am I doing wrong?

peteakalad, 2021-04-24:
Not sure as I use B1/B0, but reading the docs do you not need to simply change the A6 to A8 to transmit?
https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/Commands

RasputinBerry, 2021-04-24:
It can be so easy if you can read ;-) ... 

It works, thank you for the hint !!!!

RasputinBerry, 2021-05-06:
@peteakalad @Portisch 

This time I've read through the various documents, but can't find a solution.

Question: 
The command "cmd / bridge / Backlog RfRaw AAA806012A1A15D11E55; RfRaw 0" works in the Tasmota console.
The command "RfRaw AAA806012A1A15D11E55" also works on its own in the console.

Then it should work in MQTT.fx with "cmnd / bridge /" and with the payload "RfRaw AAA806012A1A15D11E55" and then "Publish", but it doesn't.

In the ioBroker (SONOFF Adapter 2.4.0.) I cannot find any way to issue the command.

I hope that you can help a second time

noahkamara, 2021-08-09:
For me, every command like:
`cmd/anythinghere/Backlog RfRaw AAA806012A1A15D11E55; RfRaw`
works in the console but not with mqtt.
For mqtt `cmd/tasmotas/Backlog` works for me