Portisch/RF-Bridge-EFM8BB1 #210: online converter is down

rogercrespo, 2021-12-26:
Hello,

the online converter is down https://bbconv.hrbl.pl/ is it possible to to bring it back again? I'm having difficult to use the CLI commands because I get error in some "import modules"

Thanks.

Marmoulak, 2021-12-27:
I have the same problem. please fix it

Arganox, 2021-12-28:
I don't think the website is maintained by @Portisch .
You can use the included Python script to convert and even send the data to your bridge.

Do check the PR 
#142

m4rquez, 2022-03-02:
a good easy to user online alternative
https://npm.runkit.com/b1-to-b0-convertor

caj2310, 2022-06-12:
> a good easy to user online alternative https://npm.runkit.com/b1-to-b0-convertor

Hi
How do I use this?  Let's say I want to convert: 
AA B1 04 1414 0262 0118 1450 38192A192A1A1A19292A19292A1A1929292929292929292A192A1A1929292A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A1A192A1929292A192A192A1A1929 55

What command / format do I enter.  Tried several ways but keep getting Unexpected token, expected ; (1:51)

chrismox, 2023-04-19:
> Hi How do I use this? Let's say I want to convert: AA B1 04 1414 0262 0118 1450 [...]

Try pasting this into the box and then pressing the "run" button.

var b1ToB0Convertor = require("b1-to-b0-convertor")
const convertB1toB0 = require('b1-to-b0-convertor');
console.log(convertB1toB0('AA B1 04 1414 0262 0118 1450 38192A192A1A1A19292A19292A1A1929292929292929292A192A1A1929292A1A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A1A192A1929292A192A192A1A1929 55'));