Portisch/RF-Bridge-EFM8BB1 #213: Sniffing Warema Shutter

tombrain, 2022-01-10:
### Latest firmware used?
yes

### Information
Warema EWFS based devices, e.g. shutters (encoding type: Manchester)

Always pushed the same button and nearly almost sniffed another code.

Additional info:
At the moment i use an c++-programm (https://github.com/rfkd/aircontrol) to send the command successfully. i want to port the code to rf-bridge

with aircontrol:
```
	airCommand =
		"S01110100110111"	// Command
		"S000011010"                  // Device
		"S011110101"                  // Device
		"S";
```
### Sniffed data
{"RfRaw":{"Data":"AA B1 04 0366 06E0 16E4 1A54 B0808081809190A18080809191AA180809190818091808 55"}}
{"RfRaw":{"Data":"AA B1 04 0366 06E0 16EE 1A54 B0808081809190A18080809191AA180809190818091808 55"}}
{"RfRaw":{"Data":"AA B1 04 0370 06E0 16EE 1A7C B0808081809190A18080809191AA180809190818091808 55"}}
{"RfRaw":{"Data":"AA B1 04 0384 06EA 16DA 1A68 B0808081809190A18080809191AA180809190818091808 55"}}
{"RfRaw":{"Data":"AA B1 04 0384 06EA 16E4 1A72 B0808081809190A18080809191AA180809190818091808 55"}}
{"RfRaw":{"Data":"AA B1 04 0384 06F4 16DA 1A68 B0808081809190A18080809191AA180809190818091808 55"}}
{"RfRaw":{"Data":"AA B1 04 0384 06F4 16DA 1A68 B0808081809190A18080809191AA180809190818091808 55"}}
{"RfRaw":{"Data":"AA B1 04 0384 06F4 16DA 1A68 B0808081809190A18080809191AA180809190818091808 55"}}
{"RfRaw":{"Data":"AA B1 04 0384 06F4 16DA 1A72 B0808081809190A18080809191AA180809190818091808 55"}}
{"RfRaw":{"Data":"AA B1 04 0384 06F4 16DA 1A72 B0808081809190A18080809191AA180809190818091808 55"}}
{"RfRaw":{"Data":"AA B1 04 0384 06F4 16E4 1A72 B0808081809190A18080809191AA180809190818091808 55"}}
{"RfRaw":{"Data":"AA B1 04 0384 06F4 16E4 1A72 B0808081809190A18080809191AA180809190818091808 55"}}

chiefymuc, 2022-02-24:
Let me know if you figured it out. Would also like to use a bridge rather than another Raspberry Pi!

m4rquez, 2022-03-02:
Hi @tombrain, did you find any solution yet?

I'm facing the same problem. Using Warema EWFS shutters with aircontrol without any issues, but can't get them working with the rf-bridge/tasmota/portisch.

Like you, i'm getting about 8 different codes, converted them all to B0, but none of them is working.

Another problem, only UP & STOP generating the long codes, DOWN codes are shorter and always the same.

tombrain, 2022-03-02:
> Let me know if you figured it out. Would also like to use a bridge rather than another Raspberry Pi!

Hi @chiefymuc,
exact. I want to port to a NUC. This engine has no Pins. I want to use the RF-Bridge for 433-Communication

tombrain, 2022-03-03:
> Using Warema EWFS shutters with aircontrol without any issue

Hi @m4rquez,
sorry, i found no solution to create Codes for the bridge.

Until now i have to use aircontrol.

tombrain, 2022-12-14:
Now i use an esp32 and ESPHome. Communication by MQTT to Home Assistant. That works for me.

chiefymuc, 2022-12-17:
Can you give some details? Which receiver / transmitter did you connect to the esp32? How did you get the codes? Thank you!

tombrain, 2022-12-19:
- Codes snifft by aircontrol
- Receiver & Transmitter
[image]
- Flash ESP with https://github.com/tombrain/warema_mqtt_bridge
- Send Codes from HA via mqtt

tombrain, 2022-12-19:
But next time i will do some changes to work by ESPHome. And additioinally some doku :-)

chiefymuc, 2023-01-05:
@tombrain I'm trying to recreate that. I have installed aircontrol, but how exactly did you get the codes? I am able to save a file and replay it, but I struggle how to convert them into actual codes.

tombrain, 2023-01-06:
@chiefymuc good news. i extend aircontrol to sniff waremacodes direct

https://github.com/tombrain/aircontrol branch: localwaremacodes

run with paramter x and directly get the code

[image]

chiefymuc, 2023-01-07:
Oh wow I did not see that! Amazing work!

I was looking into trying out ESPhome with the Raw Action:
https://esphome.io/components/remote_transmitter.html?highlight=rf#remote-transmitter-transmit-raw-action

Like S is +5000
0 is +890, -890
1 is -890,+890
s is -5000 (not used in Warema as it seems).

This might be a nice option as well to work with ESP32. I will see if I can find the time.

I made it work with your Arduino code on ESP32 so far!

Blackspell01, 2023-06-02:
hey there, apparently im too stupid to get it working....
whenever im trying to perform a scan with aircontrol -d scan -s 5000 im only getting a 40kb file which I can't open ... can someone please explain what I have to do?

chiefymuc, 2023-06-02:
- It's a hex file. You can upload it into hex converters online and then you should be able to see a bunch of 0 and 1. I then plotted the result as a graph and started to look where a code similar to the above started... It's tricky

Blackspell01, 2023-06-02:
thank you for your quick reply. The problem is that the file is empty except for "CE C0 AD DE " the rest are 0000... Idk what im doing wrong

Blackspell01, 2023-06-17:
> * It's a hex file. You can upload it into hex converters online and then you should be able to see a bunch of 0 and 1. I then plotted the result as a graph and started to look where a code similar to the above started... It's tricky

I figured out how to convert the file (at least I think so). Now I have tons of 0 and 1s. How do I plot this as a graph?

tombrain, 2023-06-19:
@Blackspell01 did you downlad the code from https://github.com/tombrain/aircontrol branch: localwaremacodes

Blackspell01, 2023-06-20:
> @Blackspell01 did you downlad the code from https://github.com/tombrain/aircontrol branch: localwaremacodes

Actually had to press the button a few times for the software to recognise the codes. Now it worked, thank you a 1000 times for this great software!!

chiefymuc, 2023-06-20:
> @Blackspell01 I use esphome. It works perfectly. It even works with percentages of the rollo.

How did you make it work with esphome and warema?

tombrain, 2023-06-20:
@chiefymuc i upload my esphome.yaml

https://github.com/tombrain/esphome_waremacontrol

there are two parts:
- first: warema_blind_component.h listening for MQTT an send it per RCSWitch
- second: ESPHome timebased for sendind MQTT

Blackspell01, 2023-06-21:
im sorry it may be a stupid question but what am I supposed to do with the .h file? :)

tombrain, 2023-06-21:
Parallel in Filesystem to yaml of esphome

Blackspell01, 2023-06-22:
Thank you. I think im very close now. Do you know by an chance how to deal with this mqtt error? 
My settings are: 
mqtt:
  broker:  @192.168.188.74
  username: MQTT
  password: mqtt
[image]
[image]

Blackspell01, 2023-06-25:
Do you have an Idea what possibly could be wrong? I can't even add the esp to home assistant.. 
It always times out...  
[image]

tombrain, 2023-06-25:
did you get working with sonoff by esp?

Blackspell01, 2023-06-25:
Unfortunately not. Im using a wemos d1 mini currently. The sonoff rf link was a very stupid purchase, I cant manage to get this thing working with any of my rf devices...
But may I ask you another question? As with your code it is possible to send manchester code, would it also be possible to send specifically timed codes over ESP Home like in the attached image? (Maybe with another .h file)
[image]

tombrain, 2023-06-25:
i think so. you have to port this code:
https://github.com/jacobmellin/unitec-rfsend/blob/master/unitec-rfsend.c

tombrain, 2023-06-25:
Perhaps you can try by native EspHome first:

https://esphome.io/components/remote_transmitter.html#setting-up-rf-devices
```
switch:
  - platform: template
    name: RF Power Button
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '100010000000000010111110'
          protocol: 2
          repeat:
            times: 10
            wait_time: 0s
```

tombrain, 2023-06-25:
second you can try my raw433-method

      - mqtt.publish:
          topic: waremacontrol/command/raw433
          payload: "0101010101110110110101"

Blackspell01, 2023-06-26:
I tried the native method already with no luck :(.
How did you convert the initial code: '100010000000000010111110' to the payload: "0101010101110110110101"? 
And also why do you suggest to use protocol 2 in this case?

tombrain, 2023-06-28:
both code ars only examples..

protocol 2 is from copy&paste of official doku

Blackspell01, 2023-06-28:
Hmm ok i already tried that. I somehow have to set the timings to 975us and 475us but I don't understand how to specify those values..

chiefymuc, 2023-11-26:
I was able to make this work with the pure esphome!

I had to convert the up and downs into the timing and then it will send the right command.

This is the crucial part:

```
      - remote_transmitter.transmit_raw:
          code: [5890,-1780,890,-890,890,-890,1780,-1780,1780,-890,890,-1780,890,
                -890,1780,-1780,890,-890,890,-890,6780,-890,890,-890,890,-1780,890,
                -890,890,-890,1780,-1780,890,-890,6780,-890,890,-890,890,-1780,1780,
                -890,890,-1780,1780,-1780,5890]

```

Here ist my full esphome .yaml:

```
esphome:
  name: esphome-warema

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "XXXX"
  password: "#YYYY"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Warema Fallback Hotspot"
    password: "ZZZZ"

captive_portal:

remote_transmitter:
  pin: GPIO25
  # RF uses a 100% carrier signal
  carrier_duty_percent: 100%

cover:
  - platform: time_based
    device_class: blind
    name: Garden Window Cover
    icon: "mdi:window-shutter"
    open_duration: 26s
    open_action:
      - switch.turn_on: garden_window_up
    close_duration: 23s
    close_action:
      - switch.turn_on: garden_window_down
    stop_action:
      - switch.turn_on: garden_window_stop
  - platform: time_based
    device_class: blind
    name: Side Window Cover
    icon: "mdi:window-shutter"
    open_duration: 27s
    open_action:
      - switch.turn_on: side_window_up
    close_duration: 23s
    close_action:
      - switch.turn_on: side_window_down
    stop_action:
      - switch.turn_on: side_window_stop

switch:
  - platform: template
    name: Garden Window Up
    id: garden_window_up
    icon: "mdi:window-shutter-open"
    entity_category: config
    turn_on_action:
      - remote_transmitter.transmit_raw:
          code: [5890,-1780,890,-890,890,-890,1780,-1780,1780,-890,890,-1780,890,
                -890,1780,-1780,890,-890,890,-890,6780,-890,890,-890,890,-1780,890,
                -890,890,-890,1780,-1780,890,-890,6780,-890,890,-890,890,-1780,1780,
                -890,890,-1780,1780,-1780,5890]
  - platform: template
    name: Garden Window Down
    id: garden_window_down
    icon: "mdi:window-shutter"
    entity_category: config
    turn_on_action:
      - remote_transmitter.transmit_raw:
          code: [5890,-1780,890,-890,890,-890,1780,-1780,1780,-890,890,-1780,890,
                -890,1780,-1780,890,-890,1780,-890,5890,-890,890,-890,890,-1780,
                890,-890,890,-890,1780,-1780,890,-890,6780,-890,890,-890,890,
                -1780,1780,-890,890,-1780,1780,-1780,5890]
  - platform: template
    name: Garden Window Stop
    id: garden_window_stop
    icon: "mdi:window-shutter-cog"
    entity_category: config
    turn_on_action:
      - remote_transmitter.transmit_raw:
          code: [5890,-890,890,-1780,890,-890,1780,-1780,1780,-890,890,-1780,890,
                -890,1780,-890,890,-890,890,-1780,6780,-890,890,-890,890,-1780,890,-890,
                890,-890,1780,-1780,890,-890,6780,-890,890,-890,890,-1780,1780,-890,890,
                -1780,1780,-1780,5890]
  - platform: template
    name: Side Window Up
    id: side_window_up
    icon: "mdi:window-shutter-open"
    entity_category: config
    turn_on_action:
      - remote_transmitter.transmit_raw:
          code: [5890,-1780,890,-890,890,-890,1780,-1780,1780,-890,890,-1780,
                890,-890,890,-890,890,-890,890,-890,890,-890,6780,-890,890,
                -890,890,-1780,890,-890,890,-890,1780,-1780,890,-890,6780,-890,
                890,-890,890,-1780,1780,-890,890,-1780,1780,-1780,5890]
  - platform: template
    name: Side Window Down
    id: side_window_down
    icon: "mdi:window-shutter"
    entity_category: config
    turn_on_action:
      - remote_transmitter.transmit_raw:
          code: [5890,-1780,890,-890,890,-890,1780,-1780,1780,-890,890,-1780,
                890,-890,890,-890,890,-890,890,-890,1780,-890,5890,-890,890,
                -890,890,-1780,890,-890,890,-890,1780,-1780,890,-890,6780,
                -890,890,-890,890,-1780,1780,-890,890,-1780,1780,-1780,5890]
  - platform: template
    name: Side Window Stop
    id: side_window_stop
    icon: "mdi:window-shutter-cog"
    entity_category: config
    turn_on_action:
      - remote_transmitter.transmit_raw:
          code: [5890,-890,890,-1780,890,-890,1780,-1780,1780,-890,890,-1780,
                890,-890,890,-890,1780,-890,890,-1780,6780,-890,890,-890,890,
                -1780,890,-890,890,-890,1780,-1780,890,-890,6780,-890,890,-890,
                890,-1780,1780,-890,890,-1780,1780,-1780,5890]
```

Blackspell01, 2023-11-26:
Damn that's impressive. Thank you so much.

iflowmac, 2024-05-18:
> I was able to make this work with the pure esphome!
> 
> I had to convert the up and downs into the timing and then it will send the right command.
> 
Hello chiefymuc,
how did you convert the "S01110100111111S011101001S000011111S" to the timings? by hand or had you a software? where did you get the exact µs-values from?

Blackspell01, 2024-05-18:
I did it by hand with a YouTube tutorial. Timing is always 890µs

chiefymuc, 2024-05-18:
I built an Excel-Table to convert the codes to the timestampts. It is not fully automatic... Maybe it helps:

[code.xlsx](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/15366688/code.xlsx)