Portisch/RF-Bridge-EFM8BB1 #217: I've been told by someone on a discord server that I can't use Portisch with the Sonoff RF Bridge in the white box

solomongumbie, 2022-04-11:
I've been told by someone on a discord server that I can't use Portisch with the Sonoff RF Bridge in the white box. Unfortunately that is all I can find and I already have one on the way. Is this information correct? Does it need to be one of the black box Sonoff RF bridges?

gjuhasz86, 2022-06-25:
You probably figured this out by now, but yes AFAIK that's correct. I recently got a white one (the box says R2), and since the wiki mentions an R2 v1.0 version of the board I thought it would work.
Unfortunately the white ones have an R2 v2.2 board in them that's not compatible with Portisch.

Pointed out  by jasebob ([here](https://community.home-assistant.io/t/new-sonoff-rf-bridge-board-need-flashing-help/344326/47)):
> U1 is an 8051 instruction set micro used to encode/decode RF It used to be EFM8BB1 now OB38R08A1 in V2.2, this is the change preventing Portish.

Fortunately I was able to find a place that still had one unit left with the black enclosure. It had an R2 v1.0 board in it.

Apparently there's a way to get more functionality of the R2 v2.2 as well but that requires soldering.
More info about all of the above can be found here:
* https://github.com/arendst/Tasmota/discussions/13283
* https://community.home-assistant.io/t/new-sonoff-rf-bridge-board-need-flashing-help/344326

mightymos, 2023-08-11:
This is not ready for users now but is being worked on:
https://github.com/mightymos/RF-Bridge-OB38S003