Portisch/RF-Bridge-EFM8BB1 #227: Dooya DD remote version support

gollima, 2023-03-26:
### Latest firmware used?
Be sure you tried the [latest release](https://github.com/Portisch/RF-Bridge-EFM8BB1/releases)

### Information
What device you try to sniff data from?
Dooya DD1602H remote that controls my blinds

### Sniffed data
Can’t get any data after any rfraw command to start sniffing

Probably the Dd remote version has a different protocol. How can I get data to code the protocol support?

addinsella, 2023-10-23:
Did you get anywhere with this? I'm having the same problem with a DD1800 remote. Sonoff rf-bridge with portisch firmware installed, rfraw 177, nothing registering at all when pushing buttons. Other RF remotes' signals are received.

gollima, 2023-10-23:
> Did you get anywhere with this? I'm having the same problem with a DD1800 remote. Sonoff rf-bridge with portisch firmware installed, rfraw 177, nothing registering at all when pushing buttons. Other RF remotes' signals are received.
> 
> 
> 
> Noteworthy perhaps that the red 'RF' light on the front of the rf-bridge does not come on when pushing buttons on the DD1800 remote. 

Nothing here... thinking in change the original controller on my blinds but risking losing it. Need more time to this project.

fauxqueue, 2023-10-24:
I'm having the same problem with a DD2702 remote. Nothing is registering in console and the RF light doesn't react. The remote has 15 channels. Tried them all. Nothing.

gollima, 2023-10-24:
I can try to code it but I don't know where to start to understand how it works. If anyone can share some information or knowledge to enable me to contribute I appreciate it.

fauxqueue, 2023-10-26:
I wouldn´t know where to begin. But maybe the hex file here works? https://github.com/Portisch/RF-Bridge-EFM8BB1/pull/221
It´s a different remote but might use the same protocol. If not, the process for adding a new protocol would probably be similar? I´ll try the hex file when I can and will report the results.

gollima, 2023-10-26:
Problem is reading my remote codes since the bridge don't show anything when I hit the remote button.

fauxqueue, 2023-10-27:
I can confirm that the hex file in https://github.com/Portisch/RF-Bridge-EFM8BB1/pull/221 doesn't help this problem. I was hoping that the software is telling the unit what signals to listen for and this was just a matter of telling it to listen for the right signals. If that's not the case and this is a hardware problem where the unit is simply incapable of receiving signals from this remote then I don't think this is fixable. 
The remote can be switched between bi-directional and mono-directional modes. Bi-directional is default. I tried switching to mono-directional and then my rfbridge receives the signals just fine! Unfortunately I can't pair the remote with the blinds in mono-directional mode so this doesn't help my case. I see that the Dooya remote can also be switched between modes. Maybe you can pair it with your motors in mono-directional mode?

addinsella, 2023-10-29:
For what it's worth, I also tried changing my remote (DD1800) to mono-directional and got the same results as above. Signals are received by Rf-bridge, but I cannot pair the remote with my blinds when it is in mono mode.

fauxqueue, 2023-10-31:
I confirmed with my blinds manufacturer, Brel, that the blinds will not work in mono-directional mode.
According to this random thread I found the Dooya remotes use a private protocol. A person in that thread got a step further than us and managed to sniff the signal but it was useless.
This is rather frustrating as I made my purchasing decision with the RFBridge plan in mind and now I´m stuck with dumb blinds. 

This might be an option: https://community.home-assistant.io/t/bi-directional-blinds-motor-supported/595685

I´m wondering if the RF signal can´t be spoofed, is there a way to open up the remote and solder an ESP to it to generate signals? I don´t have the skills to figure that out myself but if someone else has the knowhow

gollima, 2023-10-31:
Lol same with me!
I'm thinking in modify my blinds to bypass the activation mechanism to other one that I can work with.

Robmak73, 2023-11-01:
> Did you get anywhere with this? I'm having the same problem with a DD1800 remote. Sonoff rf-bridge with portisch firmware installed, rfraw 177, nothing registering at all when pushing buttons. Other [...]

Rfraw 177 is practically useless if you don't know how to add the code to the protocol. I have tried many times and asked for support. Never managed to get the code into the hex file. The python bo to b1 converter was not working last time I tried. 
I managed to find some rf433 remote control and relievers that are compatible with the porticsh main sensor file. Then swapped the circuit board. Now working great. Very reliable. If you can learn the signal and get a data code to assign to one of the 16 rfkeys. The bridge is brilliant in http or mqtt. Good range and replication. 
My advice is if you have to use any other rfraw than the default then don't bother. It's cheaper easier and quicker with less stress to just replace the relievers with a compatible code receiver and a remote to transmit for learning .I got some cheap rf433 sockets from wickes. 3 for a tenner with remote. All perfectly compatible with rf Bridge. I  failed integrating yale wireless alarm contact and pir sensor and Drayton central heating control. But luckily I had more luck with my broadlink rm Pro. But it only sends a learned signal. Which narrow its usage down

fauxqueue, 2024-06-13:
It´s been a while since this was active but I figured I´d share my solution since it doesn´t look like anyone found a better one. I never managed to get the sonoff bridge to work and I´m pretty sure it´s incompatible with these type of bi-directional motor blinds. So my solution was to buy a different device. This is what I got and I´m very happy with it http://www.rfxcom.com/epages/78165469.sf/en_GB/?ObjectPath=/Shops/78165469/Products/23104
Was fairly simple to set up, certainly simpler than flashing the Sonoff bridge. It has been rock solid and stable for six months now without a single issue.